KeystoneHQ/keystone3-firmware #381: sha256 hash does not match for 1.2.0 keystone3.bin

m-kubik, 2023-12-08:
FW release 1.2.0 stated hash 8fe42cfdfa80dd810acd51b1dd656cf81410373e9c892bc335d66d0e9e86ec3c
 at https://github.com/KeystoneHQ/keystone3-firmware/releases/tag/1.2.0 is not correct.

When calculating hash for keystone3.bin file locally I'm getting:                                                                                          
SHA256 (keystone3.bin) = **8153dcf2391512ecd57a9fc1727a114d829a82f86fe8f94205616fff7920855**

This does not match: 8fe42cfdfa80dd810acd51b1dd656cf81410373e9c892bc335d66d0e9e86ec3c as published

If I go into HW wallet settings and run verification from there then it shows the hash as stated for the release correctly

aaronisme, 2023-12-08:
In order to fully verify the integrity of the firmware, you can compile it from our source code and check the raw firmware file which is not compressed. We have provided the guide for this, checkout it here:  https://github.com/KeystoneHQ/keystone3-firmware/blob/release/v1.2.0/docs/verify.md

m-kubik, 2023-12-08:
thank you @aaronisme that explains it. Is it possible to add in also hash of the file itself so I can verify the hash even before uploading to HW wallet?

aaronisme, 2023-12-08:
For sure, we will provide the value in the github release page.

m-kubik, 2023-12-08:
solved

closed 2023-12-08 by m-kubik