KeystoneHQ/keystone3-firmware #422: error: ctaes.h: No such file when building keystore.c

dckc, 2023-12-21:
Using Docker, `python build.py` runs for some time getting and building stuff, but bombs out with...

```
[ 11%] Building C object CMakeFiles/mh1903.elf.dir/src/managers/power_manager.c.obj
[ 11%] Building C object CMakeFiles/mh1903.elf.dir/src/managers/screen_manager.c.obj
[ 11%] Building C object CMakeFiles/mh1903.elf.dir/src/msg/general_msg.c.obj
/root/work/src/managers/keystore.c:12:10: fatal error: ctaes.h: No such file or directory
   12 | #include "ctaes.h"
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mh1903.elf.dir/build.make:1337: CMakeFiles/mh1903.elf.dir/src/managers/keystore.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 11%] Building C object CMakeFiles/mh1903.elf.dir/src/msg/user_msg.c.obj
[ 11%] Building C object CMakeFiles/mh1903.elf.dir/src/managers/se_manager.c.obj
[ 11%] Building C object CMakeFiles/mh1903.elf.dir/src/presetting.c.obj
[ 12%] Building C object CMakeFiles/mh1903.elf.dir/src/ram/psram_heap_4.c.obj
/root/work/src/managers/fingerprint_process.c:13:10: fatal error: ctaes.h: No such file or directory
   13 | #include "ctaes.h"
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mh1903.elf.dir/build.make:1324: CMakeFiles/mh1903.elf.dir/src/managers/fingerprint_process.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:114: CMakeFiles/mh1903.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

I'm using docker for the build env. Docker is running in Ubuntu 22.04 (on PC hardware).

my clone is as of 703e885

docker build concluded with ...
```
 => => writing image sha256:103800ba852f985f487a998026a8f5e32a14521079c2237109ad74de64720428             0.0s
 => => naming to docker.io/library/keystone3-baker:local                                                 0.0s
 ```

dckc, 2023-12-21:
ctaes.h is perhaps from https://github.com/bitcoin-core/ctaes/blob/master/ctaes.h ?

I see...

```
root@01cffd331b1f:~/work# grep -r bitcoin-core .
./.gitmodules:	url = https://github.com/bitcoin-core/ctaes
```

oh... perhaps I need to get git submodules...

soralit, 2023-12-21:
Hi @dckc , please clone the repo with --recursive to get the submodules.

dckc, 2023-12-21:
having trouble getting submodules...

```
fatal: clone of 'git@github.com:KeystoneHQ/keystone3-firmware-release.git' into submodule path '/home/connolly/projects/keystone3-firmware/keystone3-firmware-release' failed
Failed to clone 'keystone3-firmware-release' a second time, aborting
```

I think I know the next move... update that url to https...

aaronisme, 2023-12-21:
@dckc Please check out the guide here: https://github.com/KeystoneHQ/keystone3-firmware/blob/release/v1.2.4/docs/verify.md#1-clone-and-prepare-the-repository

dckc, 2023-12-21:
win! thanks!

```console
$ sha256sum build/mh1903.bin 
69da29b26e84e057806b5589aee52d80057c178789bc0015060c414431be4102  build/mh1903.bin
```

closed 2023-12-21 by dckc