KeystoneHQ/keystone3-firmware #439: When Scanning a QR code from keplr, keystone hangs

omniwired, 2023-12-27:
When interacting with a smart contract, the wallet hangs and restarts itself.

## Firmware:
version 1.2.4

## Example message:
```
{
  "txBody": {
    "messages": [
      {
        "typeUrl": "/cosmwasm.wasm.v1.MsgExecuteContract",
        "value": {
          "sender": "MY WALLET ADDRESS HERE",
          "contract": "kujira1e224c8ry0nuun5expxm00hmssl8qnsjkd02ft94p3m2a33xked2qypgys3",
          "msg": "eyJkZXBvc2l0Ijp7fX0=",
          "funds": [
            {
              "denom": "ibc/295548A78785A1007F232DE286149A6FF512F180AF5657780FC89C009E2C348F",
              "amount": "500000000"
            }
          ]
        }
      }
    ],
    "memo": "",
    "timeoutHeight": "0",
    "extensionOptions": [],
    "nonCriticalExtensionOptions": []
  },
  "authInfo": {
    "signerInfos": [
      {
        "publicKey": {
          "typeUrl": "/cosmos.crypto.secp256k1.PubKey",
          "value": "THIS ONE I ALSO REMOVE"
        },
        "modeInfo": {
          "single": {
            "mode": "SIGN_MODE_DIRECT"
          }
        },
        "sequence": "6"
      }
    ],
    "fee": {
      "amount": [
        {
          "denom": "ukuji",
          "amount": "3054"
        }
      ],
      "gasLimit": "305306",
      "payer": "",
      "granter": ""
    }
  },
  "chainId": "kaiyo-1",
  "accountNumber": "88399"
}
```

## Other steps how to reproduce:

- Connect Keystone to Keplr
- Visit https://ghost.kujira.network/lend
- Try to lend or withdraw funds.
- Keplr will present a QR code to scan
- Scan the code
- See the wallet hangs or reset itself

## What should happen:
- The user can sign the contract integration.

soralit, 2023-12-27:
Thanks for reporting this. @omniwired , we have reproduced this issue.

omniwired, 2024-01-19:
is it fixed? so I can help testing or just close this ticket

soralit, 2024-03-18:
This should have been fixed so close this issue.

closed 2024-03-18 by soralit

omniwired, 2024-03-18:
yes, it's been fixed!