KeystoneHQ/keystone3-firmware #485: Solflare Solana raw message in swap confirmation

zz64, 2024-01-10:
Hi,
in SOLANA using Solflare, when i swap from token to token i get only raw message.
how can i trust transaction ?

[image]

Can i place SOLANA token address database in sdcard folder to give information to Keystone ?
like [ERC20](https://support.keyst.one/advanced-features/decode-defi-transactions)
any guide for this please ?

Thanks.

soralit, 2024-01-11:
Hey @zz64 , thanks for your feedback, currently such messages are not able to decoding and we're scheduling an improvement on Solana transaction parsing.

soralit, 2024-03-15:
Close this issue for we have recorded it into our internal channel. Really appreciate for your advice.

closed 2024-03-15 by soralit