KeystoneHQ/keystone3-firmware #574: Problem with enlarging Bitcoin QR code on the device after certain actions

calos9999, 2024-03-03:
Problem with enlarging Bitcoin QR CODE on the device after certain actions

Steps to reproduce the bug:

Assume that a wallet has already been created on the device:

Activate Bitcoin if it is not active

Click on Bitcoin

The QR of the Bitcoin address will be displayed.

You will be able to enlarge said QR with a tap on the screen

Next, press the 3 dots at the top

Press "Export xPUB"

The xPUB will be displayed in extended form

Then, go back

The previously displayed Bitcoin address will be displayed

Press on the screen to enlarge the QR

Abnormal behavior highlighted:

The device will not enlarge the QR and will not respond

Expected behavior:

The device enlarges the QR without any problems

Tested on firmware: 1.30 and 1.28

ww3512687, 2024-03-14:
We have reproduced this issue. Thanks for reporting this. @calos9999

calos9999, 2024-03-22:
> We have reproduced this issue. Thanks for reporting this. @calos9999 

Still present on 1.32

calos9999, 2024-04-07:
> We have reproduced this issue. Thanks for reporting this. @calos9999

Still present on 1.34

calos9999, 2024-05-10:
> > We have reproduced this issue. Thanks for reporting this. @calos9999
> 
> Still present on 1.34

Still present on 1.40

calos9999, 2024-05-18:
> We have reproduced this issue. Thanks for reporting this. @calos9999 

Still present on 1.42

calos9999, 2025-02-25:
> We have reproduced this issue. Thanks for reporting this. @calos9999 

Still present on 2.00
@ww3512687