KeystoneHQ/keystone3-firmware #632: suggestion : flag for silent erase of wallet

jpph, 2024-03-15:
Keystone wallet can record up to 3 wallet, and the number of registered wallet is discoverable by trying to add new wallet. So for attacker with some knowledge he can know if we have hidden wallet or not. Would be nice when we setup a wallet we can set 1 non erasable wallet and 2 erasable wallet. These  2 will be silently erased when user add new wallet.
This is just an idea , maybe there is other solution, it might do more bad than good...

ww3512687, 2024-03-16:
Thank you for this suggestion，It will be considered for potential future updates. @jpph

AtomicRPM, 2024-05-18:
> Keystone wallet can record up to 3 wallet, and the number of registered wallet is discoverable by trying to add new wallet. So for attacker with some knowledge he can know if we have hidden wallet [...]

If all three address spaces are in use an attacker would only know that the address space in the device is full, and the firmware will block any further account creation access.  This would render the device useless to an attacker/thief since the device is pin locked.  An attacker/thief's only action in this case is pin entry, which after a number of failed attempts would wipe the device.  

Wallets are accessed via unique pins, and then from there the setting options are accessible.  How do you suggest an attacker will be able to get past the pin block on power up to add a wallet in the first place, let alone determine that the address space inside the device is full?

jpph, 2024-05-18:
It is to be able to give a ''decoy'' pin to gouvrnt/home jacking. But by doing so and it they try create new wallet , they will see that other slot are taken. It would be nice to have a flag to allow ''silent erasing of wallet '' so the attacker will be able to create other wallet and think that the other slot were in fact empty.

AtomicRPM, 2024-05-18:
Unfortunately all your request/suggestion does is allow for a brute force of a pin.  Using your scenario above, the attacker/kidnapper/thief/organized crime member who managed to use a US $5.00 wrench from Harbor Freight across your jaw to get you to divulge the pin now can repeatedly brute force the device 4 times, then enter the decoy pin to reset the self destruct, and repeat.    I would rather the firmware not introduce attack vectors.