KeystoneHQ/keystone3-firmware #634: Unable to clone this repo

winksaville, 2024-03-19:
When attempting to clone I get an error "ERROR: Repository not found." followed by "fatal: Could not read from remote repository."
```
wink@imacpro clones % git clone git@github.com:KeystoneHQ/keystone3-firmware --recursive
Cloning into 'keystone3-firmware'...
remote: Enumerating objects: 19978, done.
remote: Counting objects: 100% (3401/3401), done.
remote: Compressing objects: 100% (1124/1124), done.
remote: Total 19978 (delta 2459), reused 2935 (delta 2264), pack-reused 16577
Receiving objects: 100% (19978/19978), 118.79 MiB | 22.90 MiB/s, done.
Resolving deltas: 100% (13118/13118), done.
Submodule 'external/ctaes' (https://github.com/bitcoin-core/ctaes) registered for path 'external/ctaes'
Submodule 'keystone3-firmware-release' (git@github.com:KeystoneHQ/keystone3-firmware-release.git) registered for path 'keystone3-firmware-release'
Cloning into '/Users/wink/prgs/clones/keystone3-firmware/external/ctaes'...
remote: Enumerating objects: 108, done.        
remote: Counting objects: 100% (18/18), done.        
remote: Compressing objects: 100% (14/14), done.        
remote: Total 108 (delta 8), reused 4 (delta 4), pack-reused 90        
Receiving objects: 100% (108/108), 49.22 KiB | 323.00 KiB/s, done.
Resolving deltas: 100% (53/53), done.
Cloning into '/Users/wink/prgs/clones/keystone3-firmware/keystone3-firmware-release'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:KeystoneHQ/keystone3-firmware-release.git' into submodule path '/Users/wink/prgs/clones/keystone3-firmware/keystone3-firmware-release' failed
Failed to clone 'keystone3-firmware-release'. Retry scheduled
Cloning into '/Users/wink/prgs/clones/keystone3-firmware/keystone3-firmware-release'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:KeystoneHQ/keystone3-firmware-release.git' into submodule path '/Users/wink/prgs/clones/keystone3-firmware/keystone3-firmware-release' failed
Failed to clone 'keystone3-firmware-release' a second time, aborting
```

I also tried `gh` but I get the same error.

Suggestions on how to resolve?

aaronisme, 2024-03-19:
Hi @winksaville please try this:

```
git clone https://github.com/KeystoneHQ/keystone3-firmware
git -c submodule.keystone3-firmware-release.update=none submodule update --init --recursive
```

winksaville, 2024-03-19:
TYVM @aaronisme, that "worked" at least no errors and external/ctaes has code in it now!
```
wink@imacpro clones % git clone git@github.com:KeystoneHQ/keystone3-firmware
Cloning into 'keystone3-firmware'...
remote: Enumerating objects: 19993, done.
remote: Counting objects: 100% (3506/3506), done.
remote: Compressing objects: 100% (1144/1144), done.
remote: Total 19993 (delta 2554), reused 3021 (delta 2349), pack-reused 16487
Receiving objects: 100% (19993/19993), 118.81 MiB | 18.73 MiB/s, done.
Resolving deltas: 100% (13131/13131), done.
wink@imacpro clones % cd keystone3-firmware 
wink@imacpro keystone3-firmware % ls -l external/ctaes 
total 0
wink@imacpro keystone3-firmware % git -c submodule.keystone3-firmware-release.update=none submodule update --init --recursive
Submodule 'external/ctaes' (https://github.com/bitcoin-core/ctaes) registered for path 'external/ctaes'
Submodule 'keystone3-firmware-release' (git@github.com:KeystoneHQ/keystone3-firmware-release.git) registered for path 'keystone3-firmware-release'
Cloning into '/Users/wink/prgs/clones/keystone3-firmware/external/ctaes'...
Skipping submodule 'keystone3-firmware-release'
Submodule path 'external/ctaes': checked out '8012b062ea4931f10cc2fd2075fddc3782a57ee4'
wink@imacpro keystone3-firmware % ls -l external/ctaes                                                                       
total 112
-rw-r--r--  1 wink  staff   1080 Mar 18 21:21 COPYING
-rw-r--r--  1 wink  staff   1243 Mar 18 21:21 README.md
-rw-r--r--  1 wink  staff   5338 Mar 18 21:21 bench.c
-rw-r--r--  1 wink  staff  20664 Mar 18 21:21 ctaes.c
-rw-r--r--  1 wink  staff   2845 Mar 18 21:21 ctaes.h
-rw-r--r--  1 wink  staff   8763 Mar 18 21:21 test.c
wink@imacpro keystone3-firmware % 
```

So that looks good, but build fails:
```
wink@imacpro keystone3-firmware % python3 build.py 
Traceback (most recent call last):
  File "/Users/wink/prgs/clones/keystone3-firmware/build.py", line 9, in <module>
    from scripts.read_feature_toggle import read_feature_toggle_build_cmd
  File "/Users/wink/prgs/clones/keystone3-firmware/scripts/read_feature_toggle.py", line 1, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'
wink@imacpro keystone3-firmware % 
```

Seems I'm missing 'yaml' and probably other modules, I suspect in need to setup a virtual environment, suggestions welcome.

Also, why didn't `--recursive` work?

aaronisme, 2024-03-20:
This is because `keystone-firmware-release` is the ci repo for our internal usage, so it is a private repo.

aaronisme, 2024-03-20:
If you would like to build the firmware on your own, please check this guide: https://github.com/KeystoneHQ/keystone3-firmware/blob/release/v1.3.2/docs/verify.md

winksaville, 2024-03-20:
Closing for now, TYVM!

closed 2024-03-20 by winksaville

winksaville, 2024-03-20:
I read the verify.md and it appears only firmware signed by keystone can be flashed, is that correct?

aaronisme, 2024-03-27:
Yes, Currently only the firmware signed by Keystone can be flashed. But we plan to support the self build firmware in the future.

winksaville, 2024-03-27:
@aaronisme, is there a tracking issue for that feature, I'd like to subscribe to it?

aaronisme, 2024-03-27:
I will keep you posted about this feature In this issue.

rztz, 2024-11-13:
I bought this device because it is open source. Now I find out that I can not flash my own build. If I only knew this before.