KeystoneHQ/keystone3-firmware #784: Keystone fails to sign EIP-3009 message

miohtama, 2024-04-09:
We have come across an issue that is specific to Keystone hardware wallet, and this wallet only. Currently repeatable on firmware 1.2.4 but likely all firmware affected.

- Ethereum has a signing standard called EIP-712 which is used to sign various messages
- Most popular messages are related to token swaps and transfers
- One of the implementations is EIP-3009 "transferWithAuthorization" that is implemented by USDC, many bridged tokens
- Polygon has a particular implementation of this standard for their bridged tokens called [UChildAdministrableERC20](https://polygonscan.com/address/0x4e26444aa327443e5e082fff306093490cc571af#code)

**What happens**
- Our [TradingStrategy.ai](https://tradingstrategy.ai/) is a Dapp EIP-3009 is used to to do deposits (instead of two transaction ERC-20 approve() + transferFrom(), saving one transaction)
- The deposits work on every other wallet, except with Keystone

**Root cause**
- EIP-712 message signing is a "flexible" standard and its implementations vary a bit
- Poygon's `UChildAdministrableERC20` implements this standard partially, namely having some issues related to the `chainId` and such parameters - we are not sure if these should be optionally or not
- All other wallets handle these parameters, or the lack of them, correctly
- Keystone doesn't work this particular implementation of EIP-712 message signing Polygon's `UChildAdministrableERC20` is using

We have tested
- Rabby + Keystone
- Rabby only
- Rabby + MetaMask
- TrustWallet
- Uniswap wallet
- etc.

Only Keystone has the issue.

**How to repeat**

- Have Rabby, Keysyone, some MATIC on Polygon
- Swap some MATIC to USDC.e (Polygon bridged USDC, not native one)
- [Go here and click deposit](https://tradingstrategy.ai/strategies/enzyme-polygon-eth-btc-usdc) 
- The transaction fails in the simulation before being broadcasted

[image]

[image]

`GasAbstraction: invalid signature` is the revert message from `UChildAdministrableERC20` when it tries to verify the signature signed with Keystone.

kenkunz, 2024-04-10:
Adding some additional details…

The stack trace returned from `viem` `simulateContract` call:
```
+page.svelte:1743 confirmPayment error: 7b209c13e9d4443a9426514205e4a666 ContractFunctionExecutionError: The contract function "buySharesOnBehalfUsingTransferWithAuthorizationAndTermsOfService" reverted with the following reason:
GasAbstraction: invalid signature

Contract Call:
  address:   0xb71F8d2cd7E6aC4d0C6810a293a1Ea0fCaE9d6a4
  function:  buySharesOnBehalfUsingTransferWithAuthorizationAndTermsOfService(address from, address to, uint256 value, uint256 validAfter, uint256 validBefore, bytes32 nonce, uint8 v, bytes32 r, bytes32 s, uint256 minSharesQuantity, bytes32 termsOfServiceHash, bytes termsOfServiceSignature)
  args:                                                                      (0x7612A94AafF7a552C373e3124654C1539a4486A8, 0xb71F8d2cd7E6aC4d0C6810a293a1Ea0fCaE9d6a4, 10000000, 0, 1711413754, 0x5975a814681d3e6361f3527382e2a621bddf5934d208bf79667e96f0aec781b6, 27, 0x0acc7ebbc5c7aba34a5f454da078723e2b7dcec492aa192e37cd2bf6d6db3dd3, 0x4206f8a4140786d7cc7b74f432b26dd5ca2cb3b8294a02cf64b42c17b81ab6fc, 1, 0xd57155360473926838908d76f397d2c6270237b610edf6a2b2c7172d71b001f0, 0x6d6cf5f2e3f5a5b3a940a032e7671df3e3ffbcca0aafc8ffcdb93c74d29f56f7632aa18a1185e7c70dccb2bf391c1ec735bdd6d0a560fa458ffd00bd24292bc01b)
  sender:    0x7612A94AafF7a552C373e3124654C1539a4486A8

Docs: https://viem.sh/docs/contract/simulateContract
Version: viem@2.5.0
    at getContractError (chunk-PCPNLTXG.js?v=f146a689:249:10)
    at simulateContract (chunk-PCPNLTXG.js?v=f146a689:966:11)
    at async simulateContract2 (chunk-WBKG57XC.js?v=f146a689:2502:31)
    at async confirmPayment (+page.svelte:1684:23)
```

Note that we were able to fix a similar issue with Uniswap wallet – details documented here:
https://github.com/tradingstrategy-ai/frontend/issues/713#issuecomment-2047926820

You can find the current code used to generate the EIP-712 signature here:
https://github.com/tradingstrategy-ai/frontend/blob/1d68a1d4efe2456c4808611e7cba8fad89caf66b/src/lib/eth-defi/eip-3009.ts#L73-L80

Let me know if there's any way we can further support debugging this issue.

soralit, 2024-04-11:
We're currently working on this issue @kenkunz @miohtama , thanks for reporting this.

Charon-Fan, 2024-04-11:
We have fixed the issue, and our testers have verified it. We will release the update ASAP. 🚀

closed 2024-04-29 by soralit