KeystoneHQ/keystone3-firmware #894: Custom BIP44 derivation paths

lbeder, 2024-04-16:
Hi,

Are there any plans to support custom BIP44 derivation paths? Keystone is an awesome device, but it's currently impossible to use it with custom derivation paths.

soralit, 2024-04-18:
Hi @lbeder , thanks for this suggestion. We got this idea very long ago (from the project start) and we found this functionality was quite advanced for all users. We'd like to collect more user feedback on this and determine when and how we can implement this. 
Also I'd like to learn how would you use this feature, only for bitcoin or other blockchains?

jpph, 2024-04-25:
It is indeed dangerous, will need to warn and educate user. There is use case it is mandatory to change derivation path, for example subaccounts from trezor and ledger. Without being able to change path, we can only access account 0 of trezor and ledger....

lbeder, 2024-04-27:
Hi @soralit,

Being able to use multiple paths is very useful to manage wallets that have lots of addresses in them. 

For example, instead of having only 1 path with 500 accounts in it, the user can make multiple paths for different use cases.

`m/44’/60’/X’/0` for addresses that only hold coin X 

`m/44’/60’/Y’/0` for addresses that are used for coin Y

`m/44’/60’/3’/0` only used for some other purpose

I understand the average user may not have many accounts, so they might not need this function. But it would definitely be a useful option. It could even be something only enabled in "expert mode" settings if you felt that would be more appropriate.

niveshsaharan, 2024-11-27:
This is such a dealbreaker since people like to store crypto for their family and friends. No one wants to mix coins in 1 account and that is why Ledger gives us option to create multiple accounts. Multiple accounts are just different derivation path:

For Taproot:
Account #0: m/86'/0'/0'
Account #1: m/86'/0'/1'
Account #2: m/86'/0'/2'
and so on.

Currently, Keystone is not supporting it and it's such a bummer. Because of this, we cannot even sign transactions for different accounts and the coins are just stuck there unless we get a ledger.

tomvodi, 2024-11-27:
I totally agree with @niveshsaharan. My assumption was that this is a basic feature of all hardware wallets. Now I also have to stick with Ledger which I wanted to get rid of.

@keystoneGithub what are the major problems with implementing this feature? 
As far as I see, the only problem is to show the current account on the Bitcoin main screen as well because there is not much space to show more informations.
On the other side, add/modify derivations paths are already implemented for Ethereum, so this shouldn't be much of a hurdle to adapt it for Bitcoin.

The "expert mode" switch suggested by @lbeder sounds nice. Experts may not be shocked to have more information packed on the screen.

slavik0329, 2024-11-27:
This is a hugely important feature for security and usability please add it