KeystoneHQ/keystone3-firmware #911: Dymension: can't vote, signing never finishes

omniwired, 2024-04-18:
How to reproduce:

1- Visit https://portal.dymension.xyz/dymension/governance
2- Try to vote
3- Scan QR code.
4- Press sign
5- The signing process never finishes ❌

omniwired, 2024-04-19:
hi @soralit 

The software wallet is Keplr.

The exact URL is https://portal.dymension.xyz/dymension/proposal/12.

The payload to be signed (and hence the QR code) looks like this:

I have edited out my address to preserve privacy.

```
{
  "types": {
    "EIP712Domain": [
      {
        "name": "name",
        "type": "string"
      },
      {
        "name": "version",
        "type": "string"
      },
      {
        "name": "chainId",
        "type": "uint256"
      },
      {
        "name": "verifyingContract",
        "type": "string"
      },
      {
        "name": "salt",
        "type": "string"
      }
    ],
    "Tx": [
      {
        "name": "account_number",
        "type": "string"
      },
      {
        "name": "chain_id",
        "type": "string"
      },
      {
        "name": "fee",
        "type": "Fee"
      },
      {
        "name": "memo",
        "type": "string"
      },
      {
        "name": "msgs",
        "type": "Msg[]"
      },
      {
        "name": "sequence",
        "type": "string"
      }
    ],
    "Fee": [
      {
        "name": "feePayer",
        "type": "string"
      },
      {
        "name": "amount",
        "type": "Coin[]"
      },
      {
        "name": "gas",
        "type": "string"
      }
    ],
    "Coin": [
      {
        "name": "denom",
        "type": "string"
      },
      {
        "name": "amount",
        "type": "string"
      }
    ],
    "Msg": [
      {
        "name": "type",
        "type": "string"
      },
      {
        "name": "value",
        "type": "MsgValue"
      }
    ],
    "MsgValue": [
      {
        "name": "proposal_id",
        "type": "uint64"
      },
      {
        "name": "voter",
        "type": "string"
      },
      {
        "name": "option",
        "type": "int32"
      }
    ]
  },
  "primaryType": "Tx",
  "domain": {
    "name": "Cosmos Web3",
    "version": "1.0.0",
    "chainId": 1100,
    "verifyingContract": "cosmos",
    "salt": "0"
  },
  "message": {
    "account_number": "62210",
    "chain_id": "dymension_1100-1",
    "fee": {
      "amount": [
        {
          "amount": "2593680000000000",
          "denom": "adym"
        }
      ],
      "gas": "129684",
      "feePayer": "dym19md" -> My address here
    },
    "memo": "",
    "msgs": [
      {
        "type": "cosmos-sdk/MsgVote",
        "value": {
          "proposal_id": 12,
          "voter": "dym19md", -> My address here
          "option": 1
        }
      }
    ],
    "sequence": "9"
  }
}
```

calos9999, 2024-04-20:
Try voting into https://wallet.keplr.app/chains/dymension/proposals/12 @soralit it works for me, can't vote on https://portal.dymension.xyz/dymension/proposal/12 either

soralit, 2024-04-22:
We've planed it into our current working sprint. Will post the progress here :D

soralit, 2024-04-23:
We fixed this issue within PR: https://github.com/KeystoneHQ/keystone3-firmware/pull/924. It will be launched at next release. @calos9999 and @omniwired .

calos9999, 2024-04-23:
Thanks @soralit 😉

omniwired, 2024-04-23:
thanks for the quick turnaround

soralit, 2024-04-29:
We have released v1.3.6 to fix this issue. Please forward to our official website or the release page of this repo to get the latest firmware :D.

closed 2024-04-29 by soralit