KeystoneHQ/keystone3-firmware #927: Custom entropy, ex Dices not available for 2nd wallet.

GeetaCX, 2024-04-24:
Only the first wallet let us change entropy source.

BTW the entropy form dices cannot be verified (according to Wicked) is this fixed?
https://www.youtube.com/watch?v=mVgPoQrbi7A

soralit, 2024-04-24:
Hi @GeetaCX, nice to meet you. Please refer to the tutorial if you want to use Dice Rolls Entropy: https://guide.keyst.one/docs/dice-roll#4eb25d0613054553813a7b92ac97810d . One thing I have to remind is that you have to change the entropy source every time you want to use Dice Roll because the state is not stored in the device permanently. 

Also after 12:00 the video says Keystone passed the verification I think, we use a different algorithm which is `Dice[1-6]` in the video.

GeetaCX, 2024-04-24:
I was looking for the option for 2nd and 3rd wallet, I don't see a reason why the option isn't available like on the first one. (no big deal)

soralit, 2024-04-24:
Yep, you can follow this step to create a new dice roll seed:
1. unlock your device with your first wallet.
2. go to `Device Settings` which is in the menu on the top-right corner in the home page.
3. go to `Wallet Settings`
4. go to `Add Wallet`
5. enter your pin or password of current wallet.
6. wait 5 secs then click `Got it`
7. go to `Create Wallet` then type a new pin and wallet name.
8. you will see `Backup Options` then click the menu icon on the top-right corner
9. go to `Change Entropy` then select `Dice Rolls`
10.  select the mnemonic type you need, standard or shamir backup. 

Hope this can be helpful. :D

GeetaCX, 2024-04-24:
You were right, I missed that option.

closed 2024-04-24 by GeetaCX