KeystoneHQ/keystone3-firmware #967: Request for more information about FMC tool

xrviv, 2024-04-30:
Hello, I'm Daniel and I work with walletscrutiny.com. 

Following instructions on the Keystone3 Pro project, found [here](https://github.com/KeystoneHQ/keystone3-firmware/blob/master/docs/verify.md):

It would appear that the checksums of the built file and the downloaded unsigned binary are identical:
```
------------------------
(SIGNED) Binary from Keystone Website :
b85edf5bd058028e708437c125ad3d67ab6da520b402d9aec59b47705f3a5509  keystone3.bin
------------------------
------------------------
Binary from build process:
427e41a1b0bb25d122e06884464f1cebd7b526b35c975a39151a8e6ffc2fc352  ./build/mh1903.bin
------------------------
------------------------
Unsigned Binary from Keystone Website :
Firmware checksum sha256: 427e41a1b0bb25d122e06884464f1cebd7b526b35c975a39151a8e6ffc2fc352 
You can check this value on your device.
------------------------
Unsigned .bin hash must be the same as mh1903.bin.
```

The unsigned binary has the same checksum with mh1903.bin **IF** you use fmc to compare them. 

However, I tried to run $ sha256sum keystone3-unsigned.bin and came up with a different result:

```
$ sha256sum keystone3-unsigned.bin 
28ed8ee1f31f53dbb83bc4d83d64b596b6033a13fe04ba9baae5863c85b1fb5f
```

I'm struggling to find information about the tool to check itself. Would be happy to receive guidance. Thank you.

soralit, 2024-04-30:
Hi @xrviv nice to meet you. Could you explain what FMC tool stands for? :D

soralit, 2024-04-30:
Also do you build the firmware with docker? The firmware might be different with different environment (Compiler, System, etc.).

soralit, 2024-04-30:
Hi @xrviv nice to meet you. If you have a read of the source code of the `firmware-maker` you can find that the `keystone3-unsigned.bin` is actually the compressed version of `mh1903.bin` with some headers, so what `firmware-checker` does is decompress the content then do a sha256 checksum.

xrviv, 2024-04-30:
Thank you @soralit we made a script to perform the instructions stated on the verification page:

This is a copy:

```
#!/bin/bash

### provide this script with the version without "v" and the published buildHash

version=$1

cd /tmp
git clone https://github.com/KeystoneHQ/keystone3-firmware
cd keystone3-firmware
git -c submodule.keystone3-firmware-release.update=none submodule update --init --recursive
docker build -t keystonehq/keystone3_baker:1.0.2 
git checkout tags/${version}
docker run -v $(pwd):/keystone3-firmware keystonehq/keystone3_baker:1.0.2 python3 build.py -e production
sha256sum ./build/mh1903.bin
wget https://keyst.one/contents/KeystoneFirmwareG3/v${version}/keystone3.bin
cd /tools/code/firmware-maker
cargo build --manifest-path tools/code/firmware-maker/Cargo.toml
./tools/code/firmware-maker/target/debug/fmm --source build/mh1903.bin --destination keystone3-unsigned.bin

cargo build --manifest-path tools/code/firmware-checker/Cargo.toml

./tools/code/firmware-checker/target/debug/fmc --source keystone3-unsigned.bin

echo "All builds complete."
echo "------------------------"
echo "(SIGNED) Binary from Keystone Website :"
sha256sum keystone3.bin     
echo "------------------------"

echo "------------------------"
echo "Binary from build process:"
sha256sum ./build/mh1903.bin
echo "------------------------"

echo "------------------------"
echo "Unsigned Binary from Keystone Website :"
./tools/code/firmware-checker/target/debug/fmc --source keystone3-unsigned.bin
echo "------------------------"

echo "Unsigned .bin hash must be the same as mh1903.bin."
```

xrviv, 2024-04-30:
I am now currently trying to figure out a way to manually decompress then run the sha256sum, but it's late here, so I might continue this tomorrow. I appreciate the help. FMM is the Firmware Maker, and FMC is the firmware checker.

soralit, 2024-05-06:
Hi @xrviv do you still have any trouble on this issue? :D

xrviv, 2024-05-06:
Please give us a little more time (max 4 working days) to conclude this. Thank you.

soralit, 2024-05-14:
Hi @xrviv , I have seen the latest review from your team, thanks a lot!

closed 2024-05-15 by soralit