KeystoneHQ/keystone3-firmware #1096: Seeking slip39 tests

onvej-sl, 2024-05-31:
I'm trying to implement a [recent revision](https://github.com/satoshilabs/slips/commit/8d060706b549af6443e04f55605b71f65c981663) of the [SLIP-39 specification](https://github.com/satoshilabs/slips/blob/master/slip-0039.md).

I would like to ask
  * whether you have tests covering slip39 mnemonic generation and recovery and
  * (if so) what the easiest way is to run them.

aaroninks, 2024-06-02:
Thanks for the question, 
Do you want to implement this on the Keystone3 firmware?

aaroninks, 2024-06-02:
The test vector defined in the SLIP-39 file can be helpful. https://github.com/satoshilabs/slips/blob/master/slip-0039.md#test-vectors. We are also using it to test our implementation.

onvej-sl, 2024-06-03:
> Do you want to implement this on the Keystone3 firmware?

Yes.

> We are also using it to test our implementation.

Where can I find the tests? How do I run them?

aaroninks, 2024-06-03:
> > Do you want to implement this on the Keystone3 firmware?
> 
> Yes.
> 
> > We are also using it to test our implementation.
> 
> Where can I find the tests? How do I run them?

these test cases are included in another repo which is used internally for testing purpose. The repo is not public for new.

fmunteanu, 2024-12-19:
@aaroninks I'm unable to import a Trezor backup, probably because latest SLIP-39 is not present into Keystone firmware. See https://github.com/KeystoneHQ/keystone3-firmware/issues/1506.

8de2fdb0, 2025-01-21:
@aaroninks I refactored the work from @onvej-sl on top of current master. Unfortunately I have the same issues with adding tests.

I see that there are test cases read via USB when the firmware runs on a real device, the simulator doesn't seem to have that ability. Any help with adding test cases would be appreciated.

The changes are fairly small and in theory it should be backward compatible. 
There is the odd case where the iteration exponent masks the extendable backup flag, since the iteration exponent used to be 5 bit and is now 4 bit where the first bit became the extendable backup flag. It looks as the [firmware sets  it to 0](https://github.com/onvej-sl/keystone3-firmware/blob/b0f6fa9b4b924349425e500b8d9dcd9762af46c9/src/crypto/slip39/slip39.c#L677), but if a share hat the exponent set to be big enough to make the first bit toggled to high it would look as if it is a slip39 secret that has extendable backups enabled and the decryption would get a truncated pbkdf2 count and the final seed would be wrong. But that is very unlikely.

fmunteanu, 2025-01-21:
Are the devs looking into this? This is blocker for importing Trezor SLIP-39 seed phrases, I have a difficult time why this is not addressed quickly.

dariuszsekula, 2025-08-13:
> Are the devs looking into this? This is blocker for importing Trezor SLIP-39 seed phrases, I have a difficult time why this is not addressed quickly.

Any updates to support shamir wallet and use it for example with cardano in Drapp wallet?