KeystoneHQ/keystone3-firmware #1104: Permissioned `keystone3-firmware-release` access - unable to clone

fmorency, 2024-06-07:
I cannot recursively clone the `keystone3-firmware` repository because the `keystone3-firmware-release` submodule is permissioned.

```shell
➜  dev git clone https://github.com/KeystoneHQ/keystone3-firmware --recursive
Cloning into 'keystone3-firmware'...
remote: Enumerating objects: 34717, done.
remote: Counting objects: 100% (8508/8508), done.
remote: Compressing objects: 100% (2804/2804), done.
remote: Total 34717 (delta 6346), reused 7448 (delta 5666), pack-reused 26209
Receiving objects: 100% (34717/34717), 137.87 MiB | 9.18 MiB/s, done.
Resolving deltas: 100% (24743/24743), done.
Submodule 'external/ctaes' (https://github.com/bitcoin-core/ctaes) registered for path 'external/ctaes'
Submodule 'keystone3-firmware-release' (git@github.com:KeystoneHQ/keystone3-firmware-release.git) registered for path 'keystone3-firmware-release'
Cloning into '/home/fmorency/dev/keystone3-firmware/external/ctaes'...
remote: Enumerating objects: 108, done.        
remote: Counting objects: 100% (18/18), done.        
remote: Compressing objects: 100% (14/14), done.        
remote: Total 108 (delta 8), reused 4 (delta 4), pack-reused 90        
Receiving objects: 100% (108/108), 49.22 KiB | 1008.00 KiB/s, done.
Resolving deltas: 100% (53/53), done.
Cloning into '/home/fmorency/dev/keystone3-firmware/keystone3-firmware-release'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

If I click on the submodule in GitHub I get

[image]

aaronisme, 2024-06-07:
@fmorency please check the doc for clone the project, `https://github.com/KeystoneHQ/keystone3-firmware/blob/master/docs/verify.md#detailed-steps-for-building-and-verifying-the-firmware`

aaronisme, 2024-06-07:
The submodule `keystone3-firmware-release` is our internal and private repo for building the firmware

fmorency, 2024-06-07:
Thanks, @aaronisme!

The instructions in the README should probably be updated.

closed 2024-06-11 by aaroninks