KeystoneHQ/keystone3-firmware #1272: Document and link from the README which firmware components are not open source

daira, 2024-08-17:
In a thread on the Zcash forum, Lixin [wrote](https://forum.zcashcommunity.com/t/keystone-hardware-wallet-support-grant-application/48508/52):
> Our firmware is open source under MIT license. It can be rebuilt and verified. We do rely on some lib from MCU vendor. Because this MCU has some optimization on it, which is the main reason we choose it. I don’t think we can remove those dependencies.

Please document which libraries those are and what they are used for, and link that doc from the README. I realize that MCU vendors don't make it easy to have 100% open source firmware, but this would help potential users to assess to what extent they're comfortable with having those closed-source dependencies.

aaronisme, 2024-08-17:
Hi @daira, thanks for the feedback. In software development, there are more or less external dependencies. While we aim to make everything open source, we can't mandate others to do the same. Our firmware relies on the MCU and requires some vendor libraries for quality development.

Currently, the only library we use is [QRDecodeLib](https://github.com/KeystoneHQ/keystone3-firmware/blob/master/external/mh1903_lib/MHSCPU_Driver/lib/MH1903_QRDecodeLib.a). This MCU is optimized for QR code scanning on the hardware side, which is the main reason we chose it.

A new section will be added to the readme file to explain the code structure and list the libraries used, enhancing understanding of our firmware.

aaroninks, 2024-08-17:
Hi @daira, thanks for the feedback. In software development, there are more or less external dependencies. While we aim to make everything open source, we can't mandate others to do the same. Our firmware relies on the MCU and requires some vendor libraries for quality development.

Currently, the only library we use is [QRDecodeLib](https://github.com/KeystoneHQ/keystone3-firmware/blob/master/external/mh1903_lib/MHSCPU_Driver/lib/MH1903_QRDecodeLib.a). This MCU is optimized for QR code scanning on the hardware side, which is the main reason we chose it.

A new section will be added to the readme file to explain the code structure and list the libraries used, enhancing understanding of our firmware.

kayabaNerve, 2024-08-19:
The ability to use QR codes is advertised for its safety compared to USB/BT. When the library for the QR codes isn't FOSS, and presumably isn't in a memory-safe language (I'd guess C as that's what most MCU libs I've seen use, though I'm not an expert here), those safety claims become challenged.

Are there numbers on the performance with this library, and the performance without? Would Keystone offer a variant users can install which does not have any blobs, even if significantly slower?

aaroninks, 2024-08-19:
> The ability to use QR codes is advertised for its safety compared to USB/BT. When the library for the QR codes isn't FOSS, and presumably isn't in a memory-safe language (I'd guess C as that's what [...]

It is important to note that the library's primary function is to scan and decode images into raw data contained within QR codes. Users can independently verify the data to ensure it matches the sender's information.

We tried some other open source QR code decode libraries, and they're way slower than the one from the MCU vendor. Makes sense though, since they're just using software for image processing and decoding without any hardware optimization.

Thanks for your feedback. We're considering giving users the option to choose and maybe even let them change and update the firmware themselves if they don't want to use the library. But honestly, it would take a ton of engineering work to match the current setup. We'll keep you posted!

kayabaNerve, 2024-08-19:
> Users can independently verify the data to ensure it matches the sender's information.

That's insufficient in the worst case, some form of RCE via an exploit in this library (as the QR codes are untrusted inputs).

aaroninks, 2024-08-19:
> > Users can independently verify the data to ensure it matches the sender's information.
> 
> I'd presume that's insufficient in the worst case, some form of RCE via an exploit in this library (as the QR codes are untrusted inputs), yet I'd have to check the MCU arrangement to speak on that with confidence. I believe multiple are used? What operations are done on the same chip as this library?

In our system design and threat model, the input source—whether QR Code images or other image types—is considered inherently untrusted. Regardless of whether a specific library is used, the firmware must consistently validate and verify all incoming data. If the data is determined to be illegitimate or malformed, the firmware should promptly reject it and notify the user.

aaroninks, 2024-08-19:
@daira I have added a `Code Structure` section to the README file to provide more detailed information about the external dependencies.

aaroninks, 2024-08-19:
@kayabaNerve I will close this issue as it primarily concerns documenting the external dependencies. However, our team sincerely appreciates all the comments and feedback you provided. 

Meanwhile if you have any feedback, question or anything else, free to raise an discussion [here](https://github.com/KeystoneHQ/keystone3-firmware/discussions). Let's have the discussion there.

closed 2024-08-19 by aaroninks