KeystoneHQ/keystone3-firmware #1372: UI simulator run failure

baymac, 2024-09-27:
I am using Mac M1 Pro, Mac OS 14.1, followed the docs but facing this error. Not very familiar with rust, can I please get some help to understand what is this error? Same error on branches `master` and `release/v1.7.0`.

```
ERROR: Couldn't execute `cargo metadata` with manifest "/Users/parichay/minswap/keystone3-firmware/rust/rust_c/Cargo.toml": Metadata(Output { status: ExitStatus(unix_wait_status(25856)), stdout: "", stderr: "warning: patch for `bitcoin` uses the features mechanism. default-features and features will not take effect because the patch dependency does not support this mechanism\nerror: no matching package named `cardano-serialization-lib` found\nlocation searched: https://git@github.com/KeystoneHQ/cardano-serialization-lib.git?tag=keystone-0.1.3#370350ec\nrequired by package `app_cardano v0.1.0 (/Users/parichay/minswap/keystone3-firmware/rust/apps/cardano)`\n    ... which satisfies path dependency `app_cardano` (locked to 0.1.0) of package `common_rust_c v0.1.0 (/Users/parichay/minswap/keystone3-firmware/rust/rust_c/src/common)`\n" })
ERROR: Couldn't generate bindings for /Users/parichay/minswap/keystone3-firmware/rust/rust_c.
make[2]: *** [rust/rust_c/CMakeFiles/rust_c] Error 1
make[1]: *** [rust/rust_c/CMakeFiles/rust_c.dir/all] Error 2
make: *** [all] Error 2
```

soralit, 2024-09-27:
This might be related to your network status, because this dependency is fetched from github.

baymac, 2024-09-27:
I am able to install other packages, ping github.com across multiple networks. I can try it on a different machine later to verify. Btw I am wondering if I can side load firmware into the hardware directly? It would be easier to work with.

soralit, 2024-09-27:
If you mean building firmware by yourself and load it into your device, I'm sadly to say no. The released firmwares are signed and verified by the hardware. 
And the built simulator could only run on your PC.

baymac, 2024-09-27:
I tried resolving the previous error, and that issue is now fixed. However, I’m facing new errors that I can't make sense of. I've shared the logs at [link](https://gist.github.com/baymac/cf809b3892e3364a9a24759a58358b1f) for reference. If you have a solution please let me know.

We might as well move this issue to discussions tab.

soralit, 2024-09-27:
seems you didn't get the submodule, please follow the command in this section: https://github.com/KeystoneHQ/keystone3-firmware?tab=readme-ov-file#macos