KeystoneHQ/keystone3-firmware #1401: Reproducible build script for Keystone 3 Pro 1.7.2 is failing

xrviv, 2024-10-17:
Hello team Keystone, 

Danny here from WalletScrutiny. 

We're currently trying to verify the reproducibility of the Keystone 3 Pro firmware, using this script:

```
#!/bin/bash

### provide this script with the version without "v" and the published buildHash

version=$1

cd /tmp
git clone https://github.com/KeystoneHQ/keystone3-firmware
cd keystone3-firmware
git -c submodule.keystone3-firmware-release.update=none submodule update --init --recursive
docker build -t keystonehq/keystone3_baker:1.0.2 
git checkout tags/${version}
docker run -v $(pwd):/keystone3-firmware keystonehq/keystone3_baker:1.0.2 python3 build.py -e production
sha256sum ./build/mh1903.bin
wget https://keyst.one/contents/KeystoneFirmwareG3/v${version}/keystone3.bin
cd /tools/code/firmware-maker
cargo build --manifest-path tools/code/firmware-maker/Cargo.toml
./tools/code/firmware-maker/target/debug/fmm --source build/mh1903.bin --destination keystone3-unsigned.bin

cargo build --manifest-path tools/code/firmware-checker/Cargo.toml

./tools/code/firmware-checker/target/debug/fmc --source keystone3-unsigned.bin

echo "All builds complete."
echo "------------------------"
echo "(SIGNED) Binary from Keystone Website :"
sha256sum keystone3.bin     
echo "------------------------"

echo "------------------------"
echo "Binary from build process:"
sha256sum ./build/mh1903.bin
echo "------------------------"

echo "------------------------"
echo "Unsigned Binary from Keystone Website :"
./tools/code/firmware-checker/target/debug/fmc --source keystone3-unsigned.bin
echo "------------------------"

echo "Unsigned .bin hash must be the same as mh1903.bin."

```

But it is resulting in failure:

```
$ ./scripts/test/hardware/keystone3pro.sh 1.7.2
Cloning into 'keystone3-firmware'...
remote: Enumerating objects: 44250, done.
remote: Counting objects: 100% (7032/7032), done.
remote: Compressing objects: 100% (1960/1960), done.
remote: Total 44250 (delta 5264), reused 6529 (delta 4993), pack-reused 37218 (from 1)
Receiving objects: 100% (44250/44250), 157.27 MiB | 24.12 MiB/s, done.
Resolving deltas: 100% (32279/32279), done.
Submodule 'external/ctaes' (https://github.com/bitcoin-core/ctaes) registered for path 'external/ctaes'
Submodule 'keystone3-firmware-release' (git@github.com:KeystoneHQ/keystone3-firmware-release.git) registered for path 'keystone3-firmware-release'
Cloning into '/tmp/keystone3-firmware/external/ctaes'...
Skipping submodule 'keystone3-firmware-release'
Submodule path 'external/ctaes': checked out '8012b062ea4931f10cc2fd2075fddc3782a57ee4'
"docker build" requires exactly 1 argument.
See 'docker build --help'.

Usage:  docker build [OPTIONS] PATH | URL | -
[... 513 lines cut ...]
   |     ^^^^^^^^^^ help: convert the identifier to snake case: `encode_unit`

warning: `firmware-checker` (bin "fmc") generated 12 warnings (run `cargo fix --bin "fmc"` to apply 7 suggestions)
    Finished dev [unoptimized + debuginfo] target(s) in 11.39s
thread 'main' panicked at src/main.rs:36:65:
called `Result::unwrap()` on an `Err` value: NotExist
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
All builds complete.
------------------------
(SIGNED) Binary from Keystone Website :
c4d0e15eb55b0d880b643b1e3e3be50f9198799944cffea6bba6204eef985941  keystone3.bin
------------------------
------------------------
Binary from build process:
sha256sum: ./build/mh1903.bin: No such file or directory
------------------------
------------------------
Unsigned Binary from Keystone Website :
thread 'main' panicked at src/main.rs:36:65:
called `Result::unwrap()` on an `Err` value: NotExist
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
------------------------
Unsigned .bin hash must be the same as mh1903.bin.

```

soralit, 2024-10-18:
> 212   images/coin/coinOrdi.png FINISHED 23.73647689819336 ms
sh: 1: cmake: not found
213   images/coin/coinTgd.png FINISHED 25.06256103515625 ms

Seems something wrong with cmake.

xrviv, 2024-10-18:
There was also a problem with ctaes.h not being found

```/keystone3-firmware/src/managers/fingerprint_process.c:13:10: fatal error: ctaes.h: No such file or directory
   13 | #include "ctaes.h"
      |          ^~~~~~~~~
compilation terminated.
/keystone3-firmware/src/managers/keystore.c:13:10: fatal error: ctaes.h: No such file or directory
   13 | #include "ctaes.h"
      |          ^~~~~~~~~
```

soralit, 2024-10-18:
> Submodule 'external/ctaes' (https://github.com/bitcoin-core/ctaes) registered for path 'external/ctaes'

This log indicates that the ctaes is correctly cloned, so I think maybe something wrong with the build system. 

We're trying to reproduce this on a Linux device.

xrviv, 2024-10-18:
Okay, cool, thank you for the effort. 

I did try this on our build server, which is debian based. 
And on my own machine, which is Ubuntu.

soralit, 2024-10-21:
Hi @xrviv , I just update the docker file on the master branch. Can you try build your image locally and try again on `master`? for some reason I can't upload the built one on our Linux machine, I'm looking for how to solve it. 
The new docker image should work.

soralit, 2024-10-28:
Hi @xrviv , do you still have any problem on this issue?

xrviv, 2024-10-29:
Currently offline. Weve been hit by typhoon. Will get back to you once power is restored

xrviv, 2024-10-31:
Alright, back! We appreciate the quick response. 

I've made a merge request on the successful build for 1.7.8. 
https://gitlab.com/walletscrutiny/walletScrutinyCom/-/merge_requests/864

Thank you!

closed 2024-10-31 by xrviv