KeystoneHQ/keystone3-firmware #1466: no matching package named `sui-types` found

johnosullivan, 2024-11-14:
When building the firmware locally on MacBook Pro `cargo metadata` is unable to find `sui-types`.

System Specs:

```bash
Graphics/Displays:

    Apple M2 Max:

      Chipset Model: Apple M2 Max
      Type: GPU
      Bus: Built-In
      Total Number of Cores: 38
      Vendor: Apple (0x106b)
      Metal Support: Metal 3
      Displays:
        DELL S3221QS:
          Resolution: 6720 x 3780
          UI Looks like: 3360 x 1890 @ 50.00Hz
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Rotation: Supported

Hardware:

    Hardware Overview:

      Model Name: MacBook Pro
      Model Identifier: Mac14,6
      Model Number: MNWA3LL/A
      Chip: Apple M2 Max
      Total Number of Cores: 12 (8 performance and 4 efficiency)
      Memory: 32 GB
      System Firmware Version: 11881.1.1
      OS Loader Version: 11881.1.1
      Serial Number (system): **************
      Hardware UUID: B8384A00-8637-5AA2-B203-F4D2886BFF97
      Provisioning UDID: 00006021-000C21A626F0C01E
      Activation Lock Status: Enabled
```

```bash
=============================================
--
Building firmware for dev
Building firmware type general
--
=============================================
No image changes detected, skip building images.
-- compile rust
-- build target path: /Users/johnosullivan/Development/keystone3-firmware/build/rust-builds
-- cargo build command: --release--targetthumbv7em-none-eabihf-prust_c--no-default-features--featuresfirmware
-- /Users/johnosullivan/Development/keystone3-firmware/build/rust-builds
-- Compiling for MAIN target
-- The C compiler identification is AppleClang 16.0.0.16000026
-- The CXX compiler identification is AppleClang 16.0.0.16000026
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (3.1s)
-- Generating done (0.1s)
CMake Warning:
  Manually-specified variables were not used by the project:

    RU_SUPPORT

-- Build files have been written to: /Users/johnosullivan/Development/keystone3-firmware/build
ERROR: Couldn't execute `cargo metadata` with manifest "/Users/johnosullivan/Development/keystone3-firmware/rust/rust_c/Cargo.toml": Metadata(Output { status: ExitStatus(unix_wait_status(25856)), stdout: "", stderr: "warning: patch for `bitcoin` uses the features mechanism. default-features and features will not take effect because the patch dependency does not support this mechanism\n    Updating crates.io index\n    Updating git repository `https://git@github.com/KeystoneHQ/keystone-sdk-rust.git`\nerror: no matching package named `sui-types` found\nlocation searched: https://github.com/KeystoneHQ/sui.git?tag=0.1.2#9a64a25a\nrequired by package `app_sui v0.1.0 (/Users/johnosullivan/Development/keystone3-firmware/rust/apps/sui)`\n" })
ERROR: Couldn't generate bindings for /Users/johnosullivan/Development/keystone3-firmware/rust/rust_c.
make[2]: *** [rust/CMakeFiles/rust_c] Error 1
make[1]: *** [rust/CMakeFiles/rust_c.dir/all] Error 2
make: *** [all] Error 2

.....

error: no matching package named `sui-types` found
location searched: https://github.com/KeystoneHQ/sui.git?tag=0.1.2
required by package `app_sui v0.1.0 (/Users/johnosullivan/Desktop/keystone3-firmware.nosync/rust/apps/sui)`
```

Any idea how to resolve this?

soralit, 2024-11-22:
Seems to be a network issue, maybe clean you cargo cache and install the dependencies again

aaroninks, 2025-12-29:
close this issue, since there is no input on this