KeystoneHQ/keystone3-firmware #1472: Verification of Binary Signature Missing

gernotpokorny, 2024-11-15:
**How can I verify that the `keystone3.bin` binary was actually signed by you? Where can I download the public key for signature verification?**

Currently, you only verify that the binary downloaded onto the device is the same as the one that was initially downloaded. However, this does not guarantee that the binary is legitimate, as it could have been tampered with (e.g., if the website was compromised). The use of a checksum (like SHA-256) ensures that the file hasn't been altered during the download process, but it doesn’t protect against malicious files being downloaded in the first place.

At this point, to verify the integrity of the binary, I would need to install it and compare it with my local build (created from the source code on GitHub). As a user, I would feel more confident if there was a way to independently verify that you signed the binary before I proceed with installation.

It would be very helpful if you could provide a signature verification process and make the corresponding public key available. This way, users can be certain that the binary was signed by you before installation.

mostafaadel33, 2024-12-19:
> **كيف يمكنني التأكد من أن `keystone3.bin`الملف الثنائي تم توقيعه بالفعل من قبلك؟ أين يمكنني تنزيل المفتاح العام للتحقق من التوقيع؟**
> 
> حاليًا، يمكنك فقط التحقق من أن الملف الثنائي الذي تم تنزيله على الجهاز هو نفس الملف الذي تم تنزيله في البداية. ومع ذلك، لا يضمن هذا أن الملف الثنائي شرعي، حيث من الممكن أن يكون قد تم العبث به (على سبيل المثال، إذا تم اختراق موقع الويب). يضمن استخدام مجموع التحقق (مثل SHA-256) عدم تغيير الملف أثناء عملية التنزيل، لكنه لا يحمي من تنزيل الملفات الضارة في المقام الأول.
> 
> في هذه المرحلة، للتحقق من سلامة الملف الثنائي، سأحتاج إلى تثبيته ومقارنته بالنسخة المحلية التي أنشأتها (تم إنشاؤها من الكود المصدر على GitHub). بصفتي مستخدمًا، سأشعر بمزيد من الثقة إذا كانت هناك طريقة للتحقق بشكل مستقل من أنك قمت بتوقيع الملف الثنائي قبل متابعة التثبيت.
> 
> سيكون من المفيد جدًا أن تتمكن من توفير عملية التحقق من التوقيع وإتاحة المفتاح العام المقابل. بهذه الطريقة، يمكن للمستخدمين التأكد من أن الملف الثنائي تم توقيعه من قبلك قبل التثبيت.

mostafaadel33, 2024-12-19:
https://github.com/KeystoneHQ/keystone3-firmware/issues/1472#issue-2662395500