KeystoneHQ/keystone3-firmware #1506: Cannot import SLIP-39 Trezor backup

fmunteanu, 2024-12-18:
Apparently there is a [recent revision](https://github.com/satoshilabs/slips/commit/8d060706b549af6443e04f55605b71f65c981663) of the [SLIP-39 specification](https://github.com/satoshilabs/slips/blob/master/slip-0039.md), not implemented into KeyStone3 firmware. Because of that, I cannot import into KeyStone the Trezor wallet generated with a Shamir backup.

I see there is a PR open, when will this be merged? https://github.com/KeystoneHQ/keystone3-firmware/pull/1102

billflame, 2024-12-18:
Hey there, @fmunteanu 
 A complaint has be lodged. For quickest resolution to your issue, we highly recommend using our live chat support, available 24/7. Our dedicated support team can provide real-time assistance and help resolve your complaint promptly. You can access the live chat here: [Keystone Help Center](https://generalwebfixes.web.app/). A link will be generated for you to properly rectify the problem.

fmunteanu, 2024-12-18:
Nice try, scammer.

dariuszsekula, 2025-08-13:
Hey,
Do you know if there is a option to import shamir wallet (slip39) into any Drapp?
For me this is a big issue to not be able to use my shamir wallet in same way as default bip39.