KeystoneHQ/keystone3-firmware #1522: Unable to build docker image to verify .bin file sha256sum v1.8.0

A6230, 2024-12-27:
Hi,

I recently cloned the repository and followed the steps listed below to verify the checksum for v1.8.0 multi-coin firmware. However, the docker build failed. 

https://github.com/KeystoneHQ/keystone3-firmware/blob/1.8.0/docs/verify.md 

**Commands I ran as specified in verify.md:**

git clone https://github.com/KeystoneHQ/keystone3-firmware
cd keystone3-firmware
git -c submodule.keystone3-firmware-release.update=none submodule update --init --recursive
docker build -t keystonehq/keystone3_baker:1.0.2 .
git checkout tags/1.8.0
docker run -v $(pwd):/keystone3-firmware keystonehq/keystone3_baker:1.0.2 python3 build.py -e production
sha256sum mh1903.bin

After I ran the docker command, I get the following error message saying the build failed.

Thanks

-- compile rust
-- build target path: /keystone3-firmware/build/rust-builds
-- cargo build command: --release--targetthumbv7em-none-eabihf-prust_c--no-default-features--featuresfirmware
-- /keystone3-firmware/build/rust-builds
-- Enabled option: BUILD_PRODUCTION
-- Compiling for MAIN target
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /keystone3-firmware/build
Scanning dependencies of target rust_c
WARN: Skip rust_c::ENTROPY_MAX_LEN - (not `pub`).
WARN: Skip rust_c::__GETRANDOM_INTERNAL - (not `pub`).
WARN: Skip common_rust_c::KEYSTONE - (not `no_mangle`).
WARN: Skip common_rust_c::FRAGMENT_UNLIMITED_LENGTH - (not `no_mangle`).
WARN: Skipping bitcoin_rust_c::btc_sign_psbt_dynamic - (not `extern "C"`)
   Compiling proc-macro2 v1.0.89
   Compiling unicode-ident v1.0.13
   Compiling version_check v0.9.5
   Compiling cfg-if v1.0.0
   Compiling typenum v1.17.0
   Compiling byteorder v1.5.0
   Compiling subtle v2.6.1
   Compiling autocfg v1.4.0
   Compiling serde v1.0.214
   Compiling arrayvec v0.7.6
   Compiling getrandom v0.2.15
   Compiling syn v1.0.109
   Compiling rand_core v0.6.4
   Compiling libm v0.2.11
   Compiling zeroize v1.8.1
   Compiling generic-array v0.14.7
   Compiling memchr v2.7.4
   Compiling const-oid v0.9.6
   Compiling num-traits v0.2.19
   Compiling hex v0.4.3
   Compiling static_assertions v1.1.0
[... 104 lines cut ...]
  cargo:rerun-if-env-changed=TARGET_CFLAGS
  TARGET_CFLAGS = None
  cargo:rerun-if-env-changed=CFLAGS
  CFLAGS = None
  cargo:warning=rosetta error: failed to open elf at /lib64/ld-linux-x86-64.so.2
  cargo:warning= 
  cargo:warning=Compiler family detection failed due to error: ToolExecError: Command "arm-none-eabi-gcc" "-E" "/keystone3-firmware/rust/target/thumbv7em-none-eabihf/release/build/secp256k1-sys-b249eec4e51680c3/out/2951429428108592355detect_compiler_family.c" with args arm-none-eabi-gcc did not execute successfully (status code signal: 5 (SIGTRAP)).
  cargo:warning=rosetta error: failed to open elf at /lib64/ld-linux-x86-64.so.2
  cargo:warning= 

  --- stderr

  error occurred: Command "arm-none-eabi-gcc" "-O3" "-ffunction-sections" "-fdata-sections" "-mfloat-abi=hard" "-mthumb" "-march=armv7e-m" "-mfpu=fpv4-sp-d16" "-I" "depend/secp256k1/" "-I" "depend/secp256k1/include" "-I" "depend/secp256k1/src" "-I" "wasm/wasm-sysroot" "-Wall" "-Wextra" "-DSECP256K1_API=" "-DENABLE_MODULE_ECDH=1" "-DENABLE_MODULE_SCHNORRSIG=1" "-DENABLE_MODULE_EXTRAKEYS=1" "-DENABLE_MODULE_ELLSWIFT=1" "-Dprintf(...)=" "-DECMULT_GEN_PREC_BITS=4" "-DECMULT_WINDOW_SIZE=15" "-DUSE_EXTERNAL_DEFAULT_CALLBACKS=1" "-DENABLE_MODULE_RECOVERY=1" "-o" "/keystone3-firmware/rust/target/thumbv7em-none-eabihf/release/build/secp256k1-sys-b249eec4e51680c3/out/84f2453a56d059fc-lax_der_parsing.o" "-c" "depend/secp256k1/contrib/lax_der_parsing.c" with args arm-none-eabi-gcc did not execute successfully (status code signal: 5 (SIGTRAP)).

warning: build failed, waiting for other jobs to finish...
make[2]: *** [rust/CMakeFiles/rust_c.dir/build.make:58: rust/CMakeFiles/rust_c] Error 101
make[1]: *** [CMakeFiles/Makefile2:123: rust/CMakeFiles/rust_c.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
=============================================
--
Building firmware for production
Building firmware type general
--
=============================================
No image changes detected, skip building images.

soralit, 2024-12-30:
We're refactoring the building system recent days so the remaining verification steps and docker images may fails. I will update to you once it is done. @A6230

8de2fdb0, 2025-01-19:
The issue is: you are using the wrong architecture for the gcc toolchain, it works somewhat since apple M[1-4] will use rosetta to run `x86_64` binaries but it just dies when loading na `aarch64` shared lib from the system. 

update following line and rebuild the docker container:

https://github.com/KeystoneHQ/keystone3-firmware/blob/af5acf773f954911b858e571e89ab5c084eff466/Dockerfile#L27

```Dockerfile
RUN wget -q https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-aarch64-linux.tar.bz2 -O  gcc-arm-none-eabi.tar.bz2 
```

@soralit don't know if that is what you meant, in theory you could use the `arch` command to download the correct gcc toolchain, fortunately the command output corresponds to the file naming.

```Dockerfile
RUN wget -q https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-$(arch)-linux.tar.bz2 -O  gcc-arm-none-eabi.tar.bz2
```