KeystoneHQ/keystone3-firmware #1768: Simulator signing fails

thepen2, 2025-05-24:
Tried to compile the simulator with

python3 build.py -o simulator

Gets ALL the way to the end of the compile and build and then flops signing the executable.

Signing the executable
keystone simulator: no identity found
make[2]: *** [simulator] Error 1
make[2]: *** Deleting file `simulator'
make[1]: *** [CMakeFiles/simulator.dir/all] Error 2

Any have ideas??

Same exact problem with the -t btc_only option engaged as well.

Peace,

Pen

soralit, 2025-05-26:
Hey @thepen2 seems you're building the simulator on MacOS, have you ever read this part of the documentation? https://github.com/KeystoneHQ/keystone3-firmware/blob/master/docs/SIMULATOR.md#macos

soralit, 2025-05-27:
Hey @thepen2 , I'm engineer from Keystone team. 
Thanks for your sharing.
1. I just tried to install the latest SDL2 lib and it installed version 2.32.6. Please see is there a soft link `SDL2` in `/opt/homebrew/include` point to `../Cellar/sdl2/2.32.6/include/SDL2` and another soft link `sdl2` in `/opt/homebrew/opt` link to `../Cellar/sdl2/2.32.6`, if not you can try to create these links manually (not copying the files). Or you can try to remove the SDL2 lib by `brew uninstall` and try install it again. This seems to be a brew issue and that's all the thing I know. If it still not work you can try to modify the code at https://github.com/KeystoneHQ/keystone3-firmware/blob/master/ui_simulator/simulator.cmake#L76 to point to your local SDL2 directory to see if it work. 
2. About python issue I'd suggest to use a virtual environment tools to manage your python versions. You can try `miniconda` or `pyenv`. 
3. About the simulator size issue it should be 800x480, this may related to your screen scale value I guess. the definition is at https://github.com/KeystoneHQ/keystone3-firmware/blob/master/ui_simulator/lv_drv_conf.h#L98

thepen2, 2025-05-27:
Dear soralift,

Thank you for highlighting the settings in the lv_drv_conf.h file.

Reducing the SDL_ZOOM definition number there worked for us.  But we can report that modifying the SDL_HOR_RES and SDL_VER_RES definitions did not work for us except for the very first screen, after which those numbers were not respected as to the contents of the screens.

We are actually working on something rather ambitious.  We made the project decision to debut using with a quantum resistant signing algorithm already in place, specifically Falcon 9.  I know that Algorand (which you apparently don't yet support) is using this same protocol for their "state proofs" but we have not examined their code, so we don't know if it is required for a hardware wallet or if it is just something used by their validators.

As a first step, I suppose at some point we might ask you folks to sign off on a development version firmware akin to your BTC_ONLY implementation, as we'd need to bring in a Falcon library file in C for the compile.  After you are comfortable with the integrity of the library itself we might then offer support in your main firmware.

Does that sound like a reasonable possibility?

Thank you again for your most helpful attention.

Peace,

Pen

soralit, 2025-05-28:
Hi @thepen2 , really appreciate for your sharing, however I'd sorry to say we only sign the release build from our team at the moment for some security concerns from the company point of view.